orangecrab-fpga/orangecrab-examples #1: Analog examples

jmi2k, 2020-03-29:
Provide an example using the analog pins.

gregdavill, 2020-07-11:
I will spin up a standalone example for this at some point.

But we had to complete this functionality for testing units before shipping. So in the mean-time you can see an example of how it works here. Conversion time is ~800us for the RC time constant.

Here is the firmware interface: https://github.com/gregdavill/OrangeCrab-test-sw/blob/main/fw/asense.c
And here is the gateware: https://github.com/gregdavill/OrangeCrab-test-sw/blob/main/hw/rtl/analog.py

cryo1970, 2024-05-11:
Sorry -- but I don't fully understand. Will the fpga "see" oscillations on for example A0 when an analog signal is applied? And then I have to count how many oscillations I get per time unit to estimate the voltage that is present on said pin?

cryo1970, 2024-05-13:
Okay -- I see now that my question was quite ignorant, and I'm beginning to realize, that I have to implement my own SAR ADC solution, but I still have many questions to which answers are not easily available (in my opinion)....: Which physical pins from the ECP5 are involved in "controlling" the MUX (HC4067BQ) that is onboard? How do I charge up the external RC-network (that I assume is on board -- since a fixed 800 us time constant is mentioned)? What are A4 and A5 connected to (they are nowhere to be found in the constraints file)? The migen (?) code linked to doesn't help me much, and the https://github.com/smunaut/ice40-playground/blob/icepick/projects/icepick_test/rtl/sense.v link is dead. It may seem that I'm lazy because I don't want to spend a lot of time delving into migen (if that's what it is) so I can eventually make sense of the code linked to, but I just wish someone could provide guidelines for using the analog pins pertaining to VHDL (or verilog at the very least) or maybe as a minimum some hints as to how things are physically wired on the Orange Crab board as a starting point.

gregdavill, 2024-05-14:
The ECP5 doesn't contain a traditional ADC. But the hardware has features that let you pretty reasonably take low speed ADC measurements.

The OrangeCrab adds an analog mux, and then uses a differential input on the FPGA to implement an ADC based on measuring a RC time-constant. Here is a reference from Lattice about the main idea: https://www.latticesemi.com/products/designsoftwareandip/intellectualproperty/referencedesigns/referencedesign03/simplesigmadeltaadc

The schematic shows how the ECP5 is connected upto the mux: https://github.com/orangecrab-fpga/orangecrab-hardware/blob/main/hardware/orangecrab_r0.2.1/Production/OrangeCrab_r0.2.1_sch.pdf

ADC_SENSE_H (H3), ADC_SENSE_L (G3) connect to the differential inputs.
ADC_CTRL0 (G1) connects into ADC_SENSE_L via an RC 5.1k/100nF R46/C28. This pin is driven by the FPGA and a timer started that stops when the differential input toggles.

ADC_CTRL1 (F1), ADC_MUX (F4,F3,F2,H1) control the mux selection.

cryo1970, 2024-05-15:
> Links:

Very helpful!

Thank you for this. I guess I should have been able to find that 
information myself, and I'm sorry to take your time with such trivial 
questions, but as you have most likely figured out, I'm still in early 
stages of learning.

Now I have something to go on, and I'm very pleased with that.

Thank you once again.

Kind regards,
Christian