orangecrab-fpga/orangecrab-examples #3: LiteX SoC example fails synthesis on recent Yosys

shawnanastasio, 2020-07-11:
When building the litex CircuitPython SoC example with Yosys eef0ec6a, the following error is encountered:
```
4.39. Executing DFFLEGALIZE pass (convert FFs to types supported by the target).
ERROR: Conflicting init values for signal 1'0 (\soc_basesoc_interface_adr [5] = 1'x != 1'0)
```

Not sure if this is a regression or there's an actual driver conflict. I'll try to look into it.

shawnanastasio, 2020-07-11:
It seems the breakage was caused by https://github.com/YosysHQ/yosys/commit/372521ca56c2ca27bb562098aa65b4e457959e77. Since the error also occurs on upstream LiteX targets, this issue probably belongs there.

It may be worth mentioning this and suggesting users to use an older commit until it's resolved in the README, though.

gregdavill, 2020-07-11:
Yep I can reproduce it here using Yosys `c0bcbe1f`. But working fine on a build from ~2wks ago.
Not sure if this is something that needs to be fixed in Yosys or Litex just yet.

gregdavill, 2020-07-11:
I added a note about that to the README, and also added a prebuilt branch: https://github.com/gregdavill/OrangeCrab-examples/tree/prebuilt/litex

gregdavill, 2020-07-13:
This issue has been fixed upstream in Yosys.

I've just tried rebuilding the SoC with the new version and didn't have any issues. :)

closed 2020-07-13 by gregdavill