orangecrab-fpga/orangecrab-examples #6: Verilog Example Build Error

having11, 2020-07-14:
I followed the getting started guide exactly as instructed here: https://gregdavill.github.io/OrangeCrab/r0.2/docs/getting-started/ and while building with make encountered this error: 

```
Generating RTLIL representation for module `\LUT4'.
ERROR: Assert `(unsigned)*c > (unsigned)' '' failed in ./kernel/rtlil.h:168.
make: *** [Makefile:13: blink.json] Error 1
```

I was attempting to build the verilog blink program for the r0.2 on x86_64 Windows 10. The toolchain was downloaded and the bin directory was added to the system PATH and works fine. 
Full output here:
```
> make
yosys -p "synth_ecp5 -json blink.json" blink.v

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Wolf <claire@symbioticeda.com>          |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+2406 (open-tool-forge build) (git sha1 eef0ec6a, x86_64-w64-mingw32-g++ 10.1.0 -Os)

-- Parsing `blink.v' using frontend `verilog' --

1. Executing Verilog-2005 frontend: blink.v
Parsing Verilog input from `blink.v' to AST representation.
Generating RTLIL representation for module `\top'.
Successfully finished Verilog frontend.

-- Running command `synth_ecp5 -json blink.json' --

2. Executing SYNTH_ECP5 pass.

2.1. Executing Verilog-2005 frontend: D:\Hackster Show Projects\Spotlight\Orange Crab\Code\fpga-toolchain-windows_amd64-nightly-20200712\fpga-toolchain\bin\../share/yosys/ecp5/cells_sim.v
Parsing Verilog input from `D:\Hackster Show Projects\Spotlight\Orange Crab\Code\fpga-toolchain-windows_amd64-nightly-20200712\fpga-toolchain\bin\../share/yosys/ecp5/cells_sim.v' to AST representation.
Generating RTLIL representation for module `\LUT4'.
ERROR: Assert `(unsigned)*c > (unsigned)' '' failed in ./kernel/rtlil.h:168.
make: *** [Makefile:13: blink.json] Error 1
```

gregdavill, 2020-07-14:
I've not seen this particular failure before. If you run `make` again does it produce the same exact error?

having11, 2020-07-14:
yes

gregdavill, 2020-07-14:
Are you able to try the `20200711` tools? 

Can you also list the exact command you ran to add the `bin` directory to your PATH? 
It's interesting that you have spaces in your paths, but I've just tried adding spaces to my folder names and that doesn't seem to cause issue.

having11, 2020-07-14:
what are the 20200711 tools? And to edit I went to the Environment Variables window for Windows 10 and added a new entry to the PATH. It works fine for all other paths and I tested it by running yosys -V and nextpnr-ecp5 -V which both worked. I don't think this is a PATH issue.

edbordin, 2020-07-14:
This might be relevant https://github.com/YosysHQ/yosys/issues/2118

gregdavill, 2020-07-14:
I was referring to this version of the tools: https://github.com/open-tool-forge/fpga-toolchain/releases/download/nightly-20200711/fpga-toolchain-windows_amd64-nightly-20200711.zip

Can you make sure that `D:\Hackster Show Projects\Spotlight\Orange Crab\Code\fpga-toolchain-windows_amd64-nightly-20200712\fpga-toolchain\bin` is at the top of your PATH?

daveshah1, 2020-07-14:
This should be reported and fixed upstream, but for now can you try running it in a folder path containing no spaces?

having11, 2020-07-14:
Ok. Changing to build 20200711 and a path with no spaces worked. Not sure which one solved it, but it works now.

having11, 2020-07-14:
New error: 
```
cp blink.bit blink.dfu
process_begin: CreateProcess(NULL, cp blink.bit blink.dfu, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [Makefile:22: blink.dfu] Error 2
rm blink.json blink.bit blink_out.config
```

gregdavill, 2020-07-14:
Ah, sorry I had not updated the Makefile to support unix/windows: cp/rm differences.

I've just pushed the change that fixes that.

having11, 2020-07-14:
Good news, I was able to upload the blink sketch! I guess I will close this issue now. Thanks for all the help.

closed 2020-07-14 by having11

gregdavill, 2020-07-14:
Thanks for reporting the issue! 

I did manage to reproduce what you were seeing. 
It looks like a combination of installing to a non-system drive and a installation path containing spaces.

https://github.com/YosysHQ/yosys/issues/2267