orangecrab-fpga/orangecrab-examples #7: RISC-V Blink and Button examples fail to build

having11, 2020-07-14:
When running `make all` for the blink or button examples in the riscv folder, the command `riscv64-unknown-elf-gcc` comes up as unknown. Full error below:
```
$ make all
riscv64-unknown-elf-gcc  -march=rv32i -mabi=ilp32 -Wl,-Bstatic,-T,sections.ld,--strip-debug -ffreestanding -nostdlib -I. -o blink_fw                                                                                                                                                                                         .elf start.s main.c
make: riscv64-unknown-elf-gcc: Command not found
make: *** [Makefile:12: blink_fw.elf] Error 127
```
$PATH is set correctly and is being built on Ubuntu linux 64-bit. The verilog examples compile fine.

gregdavill, 2020-07-14:
I'm still yet to write the getting started guide for playing around with the riscv examples.

You will need to install the riscv gcc toolchain. In general most people are using one packaged by sifive: https://www.sifive.com/boards/ (under: `Prebuilt RISC‑V GCC Toolchain and Emulator` section)

Once downloaded you will need to add it's `riscv64-unknown-elf-gcc-8.3.0-2019.08.0/bin` to your $PATH as well.

having11, 2020-07-14:
Ok. that worked. Sorry that I'm asking so many questions. I work for hackster.io ([profile](https://www.hackster.io/gatoninja236)) and make many getting-started guides for various boards. This time, it's a getting started guide for the Orange Crab, then I'll make a project with it. If you have any ideas for a project or how it should be implemented, I would love to hear them. This is my first experience programming an FPGA, although I am a senior getting a Bachelor's in Comp. Sci. Thank again!

closed 2020-07-14 by having11