orangecrab-fpga/orangecrab-examples #11: Add full PCF for examples

gregdavill, 2020-09-21:
The current verilog examples only contain a partial pcf,.
It will be useful to have a full pcf created as a reference for peoples projects.

hexdump, 2020-11-10:
Apologies in advance if this isn't the place for this kind of technical support—

I recently purchased a couple OrangeCrab r0.2 boards, and I'm looking to get USB serial working on them. I expect the code I have for the TinyFPGA BX should be plug-and-play with the ECP5, but I'm missing the PCF lines to connect my design to the USB pins on the OrangeCrab boards. I'm not quite sure what these lines should be, so just bumping this issue.

gregdavill, 2020-11-10:
Fine to ask for support here. 
I've just updated the pcf's in f83a725.

This is still missing the SD card pins, so I'll leave the issue open, but it now contains the USB pins: `usb_d_p`,`usb_d_n`,`usb_pullup`

I'm pretty sure code from the Bx should port across, probably just need to create some new PLL settings.