orangecrab-fpga/orangecrab-examples #13: The litex example does not work for me at HEAD

la6m, 2020-10-13:
Hi Greg,
I do not have much experience with verilog and migen/litex, so I build and tested all examples. The examples work except litex. The last working commit for litex is a750e682. At HEAD usb does not enumerate.

My build environment is Arch Linux, riscv gcc from SiFive and fpga-toolchain nightly-20201011 from open-tool-forge.

Due to my lack of experience I do not know how to proceed to fix this issue. Please let me know if you need more information. 

Michael

la6m, 2020-10-14:
I can confirm that the prebuilt file is working.

It is also working, if I rollback my branch to commit a750e68 and build from there. Unfortunately the build process fails between commit 724265e2 and 0337dbe. So I can not pinpoint the exact commit. But if I have to guess, commit 724265e2 (Update LiteDRAM ...) looks suspicious for me.

gregdavill, 2020-10-14:
Took a look at this. Looks like some recent changes to LiteDRAM/Litex are the issue. If I enable the console output over serial I get this on boot:

```console
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Oct 14 2020 23:12:50
 BIOS CRC passed (cc2d0cf0)

 Migen git sha1: 12e7ba6
 LiteX git sha1: d23b88f7

--=============== SoC ==================--
CPU:		VexRiscv @ 48MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		32KiB
SRAM:		8KiB
L2:		8KiB
SDRAM:		131072KiB 16-bit @ 192MT/s (CL-6 CWL-5)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b0: |00000000| delays: -
  m0, b1: |00000000| delays: -
  m0, b2: |00000000| delays: -
  m0, b3: |00000000| delays: -
  best: m0, b00 delays: -
  m1, b0: |00000000| delays: -
  m1, b1: |00000000| delays: -
  m1, b2: |00000000| delays: 00+-00
  m1, b3: |00000000| delays: -
  best: m1, b00 delays: -
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB   
   Read: 0x40000000-0x40200000 2MiB     
  bus errors:  128/256
  addr errors: 8192/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed
```

During the read-leveling it should pick up ~3 values that work, but it's getting 0. I'll try to track down the offending commits

la6m, 2020-10-19:
I have no idea why, but when I add the console output @HEAD, the example starts working for me.

```
diff --git a/litex/SoC-CircuitPython.py b/litex/SoC-CircuitPython.py
index ef973d3..61feeca 100755
--- a/litex/SoC-CircuitPython.py
+++ b/litex/SoC-CircuitPython.py
@@ -61,7 +61,7 @@ from valentyusb.usbcore import io as usbio
 
 # connect all remaninig GPIO pins out
 extras = [
-    ("gpio", 0, Pins("GPIO:0 GPIO:1 GPIO:5 GPIO:6 GPIO:9 GPIO:10 GPIO:11 GPIO:12 GPIO:13  GPIO:18 GPIO:19 GPIO:20 GPIO:21"), 
+    ("gpio", 0, Pins("GPIO:5 GPIO:6 GPIO:9 GPIO:10 GPIO:11 GPIO:12 GPIO:13  GPIO:18 GPIO:19 GPIO:20 GPIO:21"),
         IOStandard("LVCMOS33"), Misc("PULLMODE=DOWN")),
     ("analog", 0,
         Subsignal("mux", Pins("F4 F3 F2 H1")),
@@ -187,9 +187,10 @@ class BaseSoC(SoCCore):
         platform = orangecrab.Platform(revision=revision, device=device ,toolchain=toolchain)
 
         platform.add_extension(extras)
+        platform.add_extension(orangecrab.feather_serial)
 
         # Disconnect Serial Debug (Stub required so BIOS is kept happy)
-        kwargs['uart_name']="stub"
+        #kwargs['uart_name']="stub"
 
         # SoCCore ----------------------------------------------------------------------------------
         SoCCore.__init__(self, platform, clk_freq=sys_clk_freq, csr_data_width=32, **kwargs)
```
```        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Oct 19 2020 04:52:41
 BIOS CRC passed (509aeab6)

 Migen git sha1: 12e7ba6
 LiteX git sha1: 01528e64

--=============== SoC ==================--
CPU:            VexRiscv @ 48MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            32KiB
SRAM:           8KiB
L2:             8KiB
SDRAM:          131072KiB 16-bit @ 192Mbps/pin

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b0: |11100000| delays: 01+-01
  m0, b1: |00000000| delays: -
  m0, b2: |00000000| delays: -
  m0, b3: |00000000| delays: -
  best: m0, b00 delays: 01+-01
  m1, b0: |11100000| delays: 01+-01
  m1, b1: |00000000| delays: -
  m1, b2: |00000000| delays: -
  m1, b3: |00000000| delays: -
  best: m1, b00 delays: 01+-01
Switching SDRAM to hardware control.
Memtest at 0x40000000...
  Write: 0x40000000-0x40200000 (2/2MiB)
  Read:  0x40000000-0x40200000 (2/2MiB)
Memtest OK
Memspeed at 0x40000000...
  Write: 10MiB/s
  Read:  9MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
Booting from flash...
Copying 0x20100000 to 0x40000000 (422584 bytes)...
[########################################]
Executing booted program at 0x40000000

--============= Liftoff! ===============--
```

gregdavill, 2020-10-20:
The stub uart stream will leave the `ready` bit dangling, (zero). 
Which may result in a uart FIFO that never drains. Which in turn would result in the software locking up, waiting to send uart data.

Looks like at some point I swapped over the UART from `serial` (Using the feather_serial extension) to `stub`, without also setting `ready`.

gregdavill, 2020-10-20:
Ignore that comment, looks like that's the difference between `stream` and `stub` internally in LiteX.
https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc.py#L1069
```python
        # Stub / Stream
        if name in ["stub", "stream"]:
            self.submodules.uart = uart.UART(tx_fifo_depth=0, rx_fifo_depth=0)
            if name == "stub":
                self.comb += self.uart.sink.ready.eq(1)
```

So there must be something else causing an issue with the `stub` uart.

gregdavill, 2020-10-20:
So this issue self-closed, because I mentioned fixed in the commit, but I just tried the bitstream generated by a Github Action and it's not working like my locally built one.

gregdavill, 2020-10-20:
After a bit more poking around it looks like this might be overlapping with an error in the litedram `ecp5ddrphy` https://github.com/enjoy-digital/litedram/pull/223

gregdavill, 2020-10-20:
@la6m Can you try building using the latest commit, 5cf710a?
The CI build is now working for me: https://github.com/gregdavill/OrangeCrab-examples/actions/runs/318680965

azzeloof, 2020-10-21:
New commit fixes the problem for me! Thanks Greg

la6m, 2020-10-21:
Yeah, the issue is fixed for me.
Thanks.

commit 088baaab (2020-10-20): litex: Fix SoC hanging on UART stub

In LiteX a 'stub' uart is used to create the CPU interface for a UART
but detach it from any physical interface. It's created with a stream
interface with the transmit path's ready sigal fixed HIGH. However
there is a bug, it's the rx path's ready signal that is set. This
causes the UART to stall the SoC, because it is blocked from sending
data.

Fixes #13
closed 2020-10-21 by la6m