orangecrab-fpga/orangecrab-examples #19: Orange Crab r0.2 2019-12-08 won't enter bootloader mode.

TalosThoren, 2021-03-12:
I've been searching the internet to find a more appropriate place to report this, but I think this is probably the best place I've found, so opening an issue.

Just received my board from GroupGet and I can't enter DFU mode. I've confirmed my cabling is correct, I've tested the button with a multi-meter and it is working correctly.

When I hold down btn0 while plugging in the device I receive a solid green light and a solid white light, no different than when I plug in the device without holding the button down. I've confirmed my USB cables are data cables and can be handle serial communications with multiple micropython devices.

I am trying to figure out how to flash my own bootloader onto it, now, but the documentation outside of dfu-util use is sparse. All official documentation says to enter bootloader mode in this manner. I've watched video of people booting into bootloader mode and I'm certain I'm doing it correctly.

Am I missing something?

gregdavill, 2021-03-13:
This is probably the right place for this question.

It's very interesting that you can not enter DFU mode, as this is a step performed during the testing after manufacture.
It should just be as simple as holding the button while plugging it in.

As for loading your own bootloader, this is possible. But will require an external JTAG programmer. I like to use FT2232H based ones, since they're compatible with `ecpprog` a tool that can write a file directly to tho FLASH on the OrangeCrab. Alternatively I think `openFPGAloader` supports some other tools, and you can get FLASH programming to work with `openOCD` but it's just a bit slower.
You're right the basic documentation doesn't cover how to do this, that's something that should be improved.

TalosThoren, 2021-03-13:
It's definitely not booting into bootloader mode. I tried probably 100 times or more at this stage. I even tried shorting the button contacts with tweezer in case it was a problem with the button soldering, though I've since confirmed the button works with a multimeter. It's extraordinarily frustrating, quite frankly.

I've presently got partway through setup of OpenOCD and can detect the ECP5 on it, but I'm stuck on the rest of the config files needed to actually perform useful actions on the device. Using a Segger J-Link EDU mini as a generic jtag programmer. I'm trying to figure out how to configure the "target" and "board."

I was about to step away from that and attempt to use ecpprog with an FTDI instead to reflash the bootloader, but I haven't tinkered with that quite yet.

Meanwhile, I ran into errors when attempting to build the foboot bootloader itself. Trying to track down the exact script that failed here:

```0 warnings, 1 error

Info: Program finished normally.
Traceback (most recent call last):
  File "/home/dhayden/src/github.com/gregdavill/foboot/hw/./foboot-bitstream.py", line 476, in <module>
    main()
  File "/home/dhayden/src/github.com/gregdavill/foboot/hw/./foboot-bitstream.py", line 449, in main
    vns = builder.build()
  File "/home/dhayden/src/github.com/gregdavill/foboot/hw/deps/litex/litex/soc/integration/builder.py", line 184, in build
    vns = self.soc.build(build_dir=os.path.join(self.output_dir, "gateware"),
  File "/home/dhayden/src/github.com/gregdavill/foboot/hw/deps/litex/litex/soc/integration/soc_core.py", line 446, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/dhayden/src/github.com/gregdavill/foboot/hw/deps/litex/litex/build/lattice/platform.py", line 33, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/dhayden/src/github.com/gregdavill/foboot/hw/deps/litex/litex/build/lattice/icestorm.py", line 185, in build
    _run_script(script)
  File "/home/dhayden/src/github.com/gregdavill/foboot/hw/deps/litex/litex/build/lattice/icestorm.py", line 65, in _run_script
    raise OSError("Subprocess failed")
OSError: Subprocess failed
```

I'm unclear on what subprocess is failing here, exactly. I've been tracing it through the code but can't quite figure out what the `script` is getting set to in the `if subprocess.call(shell+[script]) != 0:` call that's failing. I'm about to change the `subprocess.call` calls to `subpracess.check_output` to get a bit more insight, but I need to take a break after this update so... future David will do that.

I seem to be having trouble with this entire toolchain. Any guidance you could provide on how to actually get this device functional would be awesome. Been saving up for a bit to get it. The device itself seems physically fine, as much as I know how to troubleshoot seems okay, I feel like I just got a bum bootloader flash or something.

TalosThoren, 2021-03-13:
I just realized that for the bootloader I was not checking out the OrangeCrab branch correctly, so I'm working on building that now.

TalosThoren, 2021-04-12:
Thanks for looking at this. I ended up returning it to GroupGets for a replacement. They're telling me they mistakenly sent me an unflashed unit.

I now have the programmer you recommended in this thread, so I should be set if I end up having any further issues.

Much obliged.

closed 2021-04-12 by TalosThoren

petegeste, 2022-01-01:
@TalosThoren @gregdavill, what's the trick to load a bootloader through JTAG? I think I have a similar problem. (OC doesn't enter DFU when I connect power while holding btn0)

gregdavill, 2022-01-01:
If you'd like to reflash, you can use this pre-built bitstream: https://github.com/gregdavill/OrangeCrab-test-sw/blob/main/prebuilt/foboot-v3.1-orangecrab-r0.2-25F.bit

If you have an FT232H/FT2232H based JTAG adapter then you can make use of ecpprog: https://github.com/gregdavill/ecpprog which is included in the oss-cad-suite tools.
```
$ ecpprog foboot-v3.1-orangecrab-r0.2-25F.bit
  init..
  IDCODE: 0x41111043 (LFE5U-25)
  ECP5 Status Register: 0x00200000
  reset..
  flash ID: 0xEF 0x40 0x18 0x00
...
```

If you have a different JTAG adapter you can also make use of openOCD or openFPGAloader