orangecrab-fpga/orangecrab-examples #20: PCF files do not match the actual pinout for GPIOs

jeremyherbert, 2021-07-05:
It appears that the GPIO pins in the pcf file here: https://github.com/gregdavill/OrangeCrab-examples/blob/main/verilog/orangecrab_r0.2.pcf are not correct.
 
They do not match the pinout here: https://gregdavill.github.io/OrangeCrab/r0.2/docs/pinout/ and at least from my quick look at the schematic, the picture is the correct reference.

jeremyherbert, 2021-07-06:
I had a closer look at this, it seems that the pinout numbers on the silkscreen do not match the GPIO numbering; is this intended?

umarcor, 2021-07-07:
@jeremyherbert, my perception is that the constraints file is correct, in the sense that it responds to some criteria; however, it the criteria is not coherent with the silkscreen. I would propose the following modifications to the names in the constraints file:

| constraint | silkscreen | pin | proposal |
|---|---|---|---|
| GPIO0[0] | 0 | N17 | GPIO_0 |
| GPIO0[1] | 1 | M18 | GPIO_1 |
| GPIO0[2] | 5 | B10 | GPIO_5 |
| GPIO0[3] | 6 | B9 | GPIO_6 |
| GPIO0[4] | 9 | C8 | GPIO_9 |
| GPIO0[5] | 10 | B8 | GPIO_10 |
| GPIO0[6] | 11 | A8 | GPIO_11 |
| GPIO0[7] | 12 | H2 | GPIO_12 |
| GPIO0[8] | 13 | J2 | GPIO_13 |
| GPIO0[9] | A0 | L4 | GPIO_A0 |
| GPIO0[10] | A1 | N3 | GPIO_A1 |
| GPIO0[11] | A2 | N4 | GPIO_A2 |
| GPIO0[12] | A3 | H4 | GPIO_A3 |

gregdavill, 2021-07-07:
The pinouts on the silkscreen are following the standard 'feather' numbering, which I think has some historical reasons for the numbering. 

As @umarcor has pointed out, the GPIO numbers line up, but they have gaps.
I think we should update the pcf to cover the numbers actually present on the silkscreen.

closed 2021-07-08 by jeremyherbert