orangecrab-fpga/orangecrab-examples #27: JSON backend fails on verilog blink

shingarov, 2021-12-25:
With today's tip of main branch and today's pack of the OSS CAD Suite, the simplest example -- verilog/blink -- fails to build for me, yielding the following error:
```
~/orangecrab-examples/verilog/blink $ source ~/oss-cad-suite/environment
(OSS CAD Suite) ~/orangecrab-examples/verilog/blink $ make
yosys -p "synth_ecp5 -json blink.json" blink.v
...
...
2.49. Printing statistics.

=== top ===

   Number of wires:                 10
   Number of wire bits:            119
   Number of public wires:           5
   Number of public wire bits:      31
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              2
   Number of cells:                  3
     $add                            1
     $not                            2

2.50. Executing CHECK pass (checking for obvious problems).
Checking module top...
Found and reported 0 problems.

2.51. Executing JSON backend.
ERROR: Module top contains processes, which are not supported by JSON backend (run `proc` first).
make: *** [Makefile:32: blink.json] Error 1
rm blink.json
```
Am I missing something obvious?

gregdavill, 2021-12-26:
Can you retry with the latest commit in this repo https://github.com/orangecrab-fpga/orangecrab-examples/commit/32a8c075bbcdb2d8bb7da99e4cde6d9997d88463? 

Should be fixed now. :)

shingarov, 2021-12-26:
Yes, `32a8c07` does fix the problem.  Thanks!!

closed 2021-12-26 by shingarov