orangecrab-fpga/orangecrab-examples #29: pwm_rainbow make fails

TrailBee47, 2022-09-17:
Hi,
Sorry if I am putting it here on github.
I have just got started with FPGAs and I got this kit. now everything works fine I am able to build blink and test it on device but when I try to make the pwm_rainbow example the build fails  when I run 'make'

`-- Executing script file `pwm_rainbow.ys' --
ERROR: Can't open script file `pwm_rainbow.ys' for reading: No such file or directory`

gregdavill, 2022-09-17:
Hi @TrailBee47, Which platform are you trying to run this on?

gregdavill, 2022-09-17:
Another thing to try is to run `make pwm_rainbow.ys` to execute the commands which generate pwm_rainbow.ys. This might highlight the underlying issue.

TrailBee47, 2022-09-17:
Hi @gregdavill , I am running this on Mac OS.

when I run your command it says this 
```
 make pwm_rainbow.ys
make: `pwm_rainbow.ys' is up to date.
```

kjcolley7, 2023-01-14:
For reference, I hit this too. The problem is because `make` on macOS is BSD make, which doesn't support the `$(file)` command as it's used. The solution is to instead use GNU make, through `brew install make`. This adds the `gmake` command to your `$PATH`, so you can do `gmake dfu`. Alternatively, you can add this to your `~/.bash_profile`:

```bash
alias make='gmake'
```

@TrailBee47

closed 2023-07-08 by gregdavill