orangecrab-fpga/orangecrab-examples #32: Running python3 SoC-CircuitPython.py fails

JamesTimothyMeech, 2023-06-08:
When I run `python3 SoC-CircuitPython.py` I get this error: 
`jamesmeech@Jamess-MacBook-Pro-10 litex % python3 SoC-CircuitPython.py           
lxbuildenv: v2020.6.1.1 (run SoC-CircuitPython.py --lx-help for help)
Traceback (most recent call last):
  File "/Users/jamesmeech/Desktop/Casino/orangecrab-examples/litex/SoC-CircuitPython.py", line 342, in <module>
    main()
  File "/Users/jamesmeech/Desktop/Casino/orangecrab-examples/litex/SoC-CircuitPython.py", line 306, in main
    soc = BaseSoC(toolchain=args.toolchain, sys_clk_freq=int(float(args.sys_clk_freq)),**argdict(args))
                                                                                         ^^^^^^^^^^^^^
  File "/Users/jamesmeech/Desktop/Casino/orangecrab-examples/litex/SoC-CircuitPython.py", line 334, in argdict
    r = soc_sdram_argdict(args)
        ^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jamesmeech/Desktop/Casino/orangecrab-examples/litex/deps/litex/litex/soc/integration/soc_sdram.py", line 69, in soc_sdram_argdict
    r = soc_core_argdict(args)
        ^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jamesmeech/Desktop/Casino/orangecrab-examples/litex/deps/litex/litex/soc/integration/soc_core.py", line 320, in soc_core_argdict
    for a in inspect.getargspec(SoCCore.__init__).args:
             ^^^^^^^^^^^^^^^^^^
AttributeError: module 'inspect' has no attribute 'getargspec'. Did you mean: 'getargs'?`

I assume that this means I should be using an earlier version of python. Which version should I be using?

gregdavill, 2023-07-08:
Which version of python are you using now? Ideally this should be updated, rather than you needing to downgrade.

mangelajo, 2024-02-10:
I think I am close to get it working, but I may need some help with a last bit:
https://github.com/orangecrab-fpga/orangecrab-examples/pull/36