orangecrab-fpga/orangecrab-examples #39: Unable to enter DFU mode using a GPIO pin

tuzz, 2024-07-27:
Hello, I'm fairly new to FPGA development and I apologise if this is an obvious mistake.

I'm trying to get the Orangecrab to enter DFU mode when I press an external momentary switch that I have wired up on a breadboard. I have connected 3.3V from the Orangecrab through a switch and into GND via a 10k ohm resistor. I then adapted the `blink_reset.v` example to read `input gpio_10` and assign `reset_sr` low when the button is pressed.

When I press and hold the button, it the onboard RGB LEDs are high (perhaps half brightness), but the device doesn't enter DFU mode. The LEDs stop flashing during this time which leads me to think the device is constant resetting? I've tried debouncing the switch but that didn't seem to work. I'd appreciate any help you can provide. Thank you.

```verilog
// blink_reset.v - modified
module top (
    input clk48,

    output rgb_led0_r,
    output rgb_led0_g,
    output rgb_led0_b,

    output rst_n,
    input usr_btn,
    input gpio_10
);
    // Create a 27 bit register
    reg [26:0] counter = 0;

    // Every positive edge increment register by 1
    always @(posedge clk48) begin
        counter <= counter + 1;
    end

    // Output inverted values of counter onto LEDs
    assign rgb_led0_r = ~counter[24];
    assign rgb_led0_g = ~counter[25];
    assign rgb_led0_b = 1;

    // Reset logic on button press.
    // this will enter the bootloader
    reg reset_sr = 1'b1;
    always @(posedge clk48) begin
        reset_sr <= {~gpio_10};
    end
    assign rst_n = reset_sr;

endmodule
```

gregdavill, 2024-08-08:
The bootloader in the orangecrab makes use of the ECP5's "multi-bitstream" feature. On POR the bootloader image is loaded. This checks the state of btn0, the button on the main board. If it's held then the bootloader starts up the USB and waits for an image. However if the button isn't pushed it reloads the FPGA into the users bitstream. 

From your logic you are likely correctly swapping back to the bootloader image, but the bootloader is then jumping back to your code because btn0 isn't pressed.

As a reference here is the bootlaoder code: https://github.com/gregdavill/foboot/tree/OrangeCrab

tuzz, 2024-08-08:
I see, thank you. So does that mean there is no way to enter the bootloader from a GPIO pin signal, rather than using that exact physical button (`btn0`) on the board? I was hoping to remotely program and reprogram the Orangecrab for a project but that won't work if I have to be physically present to hold the button down. Thanks again.

tuzz, 2024-08-08:
Thanks for the help, I will take a look and report back with my findings. I will close this now. Cheers.

closed 2024-08-08 by tuzz

tuzz, 2024-12-08:
Just a quick update for posterity.

I managed to get this working using the GPIO pins on my Raspberry Pi and openFPGALoader. I soldered some header pins into the JTAG connectors on the Orangecrab and bent into rows on my breadboard. I wrote [this script](https://gist.github.com/tuzz/4ffe7c187c5ed118f8e3ed397a3fc8b8) to bootstrap my Raspberry Pi which also synthesizes and programs the FPGA.

[image]